syssi/esphome-zb-gw03 #2: What does this do exactly?

Fusseldieb, 2021-11-30:
I bought one of the GW03 hubs and came over to this repo from the HA community.
However, it's really unclear what this does.
A short description in the readme would be great!

syssi, 2021-11-30:
This [project](https://github.com/thegroove/esphome-zbbridge) is very similar for another hardware (Sonoff Zigbee Bridge). The idea is the same: Exposing the Zigbee module via serial-over-tcp.

Build blocks of the Sonoff Zigbee bridge: ESP8266 (Wifi chip) + EFR32MG21 (Zigbee module)
Build blocks of the Ewelink ZB-GW03: ESP32 (Wifi+BLE) + EFR32MG21 (Zigbee module) + LAN8720 (Ethernet module)

Fusseldieb, 2021-11-30:
That's cool! Now it makes more sense! 

> I assume you are new to esphome

Actually not, I have more than 30 ESPHome devices thoughout my home already and all of them work just beautiful.
But it's nice of you to detail it so well, it will certainly be useful for anyone that searches for it (even for me!)

> The Zigbee module is accesible via Ethernet/Telnet and is compatible with the ZHA component of HA if you make sure the correct firmware (ncp-uart-sw_6.7.8_115200.ota) is flashed.

Since it uses ncp-uart-sw, afaik it't the same Tasmots uses, isn't it? This would mean it doesn't work with multiple hub devices together, or does it? Would be lovely to mesh multiple devices together for some of them to act as just routers. The majority of the lamps I have here are **terrible** routers and are very weak too.

syssi, 2021-11-30:
I've reacted to your question at the HA community already: Are you sure a zigbee network can have multiple coordinators? AFAIK there is only **one** coordinator per network.

You are right: This component is just a network bridge between ZHA and the Zigbee module (+ zigbee firmware). The mentioned firmware is the one from the tasmota repository. I want to try these firmware versions some day:

https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle

But it is the same: The magic happens at the zigbee firmware & ZHA stack (in Home Assistant). This esphome configuration provides just the glue.

Fusseldieb, 2021-11-30:
> AFAIK there is only **one** coordinator per network.

Yep, there can be only one coordinator. That's exactly my question: If it could be reconfigured as a **router**.

> The mentioned firmware is the one from the tasmota repository.

Then afaik it can't

> But it is the same: The magic happens at the zigbee firmware & ZHA stack (in Home Assistant). This esphome configuration provides just the glue.

Ah yes! I don't quite like Tasmota, since the configurations are stored separately and not compiled. I prefer ESPHome much more....

Hedda, 2021-12-07:
> > But it is the same: The magic happens at the zigbee firmware & ZHA stack (in Home Assistant). This esphome configuration provides just the glue.
> 
> Ah yes! I don't quite like Tasmota, since the configurations are stored separately and not compiled. I prefer ESPHome much more....

FYI, ZHA is also relatively easy to get started with integrated components -> https://www.home-assistant.io/integrations/zha

syssi, 2021-12-12:
I've added a https://github.com/syssi/esphome-zb-gw03/blob/main/docs/flashing.md guide to improve the usability of this project.

closed 2023-06-25 by syssi