syssi/esphome-zb-gw03 #5: [REQUEST] Custom name in Zeroconf DNS records for unique or generic automatic network discovery by ZHA

Hedda, 2021-12-07:
Mention this here as I know you already know that ZHA support network discovery (auto-detection) can work via Zeroconf.

What is missing is the use custom name in Zeroconf DNS record for unique or generic automatic network discovery by ZHA.

https://www.home-assistant.io/integrations/zha#discovery-via-usb-or-zeroconf

More information and discussion on the Zeroconf discovery idea concept for ZHA here:

https://community.home-assistant.io/t/zha-automatic-network-discovery-of-zigbee-coordinator-bridges-gateways-ethernet-wifi-network-adapters-that-support-zeroconf-or-ssdp/293300/

As suggested there. Maybe change it to new a general “ezsp” prefix or suffix name for generic ESPHome based Zigbee remote adapters using Silabs Zigbee radios, (and respectively “znp” as a new general prefix or suffix names for generic ESPHome based Zigbee remote adapters using Texas Instruments Zigbee radios).

Using "efr32" as the prefix or suffix name will not work in the long run as thinking a bit further ahead into the future since radio-adapters for the upcoming [Matter over Thread](https://community.home-assistant.io/t/chip-matter-support-including-thread/305633) radio adapters (with devices based on Matter over Thread becoming available next year) as those will also be based on the exact same Silicon Labs “EFR32” multi-protocol MCU and radio modules, incusing the EFR32MG21 chip, because the same models Silabs EFR32 chips also support both Zigbee and Thread (as well as other IEEE 802.15.4 based specifications too).

Therefor I think using “ezsp” instead should work since Thread radio-adapters for Matter will use OpenThread firmware that uses the the [Spinel](https://openthread.io/platforms/co-processor#spinel_protocol) protocol instead, however, if would like to be absolutely sure that only adapters that have Zigbee firmware flashed then might need to add more DNS records in Zeroconf mDNS to for specifying that it specifically is a “Zigbee” radio adapter(?), perhaps by utilizing the new ZeroconfServiceInfo feature in ZHA?

https://github.com/home-assistant/core/pull/60266

https://github.com/home-assistant/core/pull/60206

https://github.com/home-assistant/architecture/discussions/662

By the way, @thegroove also started working on an updated variant with Zeroconf support for ZHA automatic discovery:

https://github.com/thegroove/esphome-zbbridge/issues/1

https://github.com/thegroove/esphome-zha-ezsp-zeroconf

https://github.com/thegroove/esphome-zeroconf

https://github.com/thegroove/esphome-zeroconf/issues

As an example see this draft WIP PR posted here:

https://github.com/home-assistant/core/pull/58224

For the ZHA side only two files updated compared to in the dev branch of Home Assistant core:

Only two files updated compared to in the dev branch of Home Assistant core:

[https://github.com/Hedda/home-assistant/blob/bd3867159e8be82aad1ead38940d26d45472b143/homeassistant/generated/zeroconf.py#L79](https://github.com/Hedda/home-assistant/blob/patch-7/homeassistant/generated/zeroconf.py#L79)

```
    "_esphomelib._tcp.local.": [
        {
            "domain": "esphome"
        },
        {
            "domain": "zha",
            "name": "tube*"
        }
    ],
    "_esphome_ezsp_bridge._tcp.local.": [
        {
            "domain": "zha",
            "name": "esphome_ezsp_bridge*"
        }
    ],
```
[https://github.com/Hedda/home-assistant/blob/bd3867159e8be82aad1ead38940d26d45472b143/homeassistant/components/zha/manifest.json#L25](https://github.com/Hedda/home-assistant/blob/patch-7/homeassistant/components/zha/manifest.json#L25)

```
"zeroconf": [
    {
      "type": "_esphomelib._tcp.local.",
      "name": "tube*"
    },
    {
      "type": "_esphome_ezsp_bridge._tcp.local.",
      "name": "esphome_ezsp_bridge*"
    }
  ],
```

Hedda, 2021-12-17:
Suggest use a generic esphome node name like "esphome_zigbee_ezsp_bridge" and submit a PR for general "`*zigbee*ezsp*`"?

https://github.com/home-assistant/core/blob/dev/homeassistant/generated/zeroconf.py#L86-L94

https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/manifest.json#L27-L31

Hedda, 2022-02-03:
FYI, Zero-configuration networking (zeroconf) looks to have been changed and improved in the Home Assistant 2022.02 release:

https://github.com/home-assistant/core/pull/62133

https://www.home-assistant.io/integrations/zeroconf/

This change is only a concern for custom integration developers.

Currently zeroconf matching only allows matching the `macaddress`, `model`, and `manufacturer` properties along with the name from the ZeroconfServiceInfo.

Since properties are arbitrarily defined by the zeroconf service, the list of named properties has grown over time.

Matching now allows for any arbitrarily defined property. All property matches must be lowercase, wildcards are supported

The top level keys `model`, `manufacturer`, and `macaddress` are now deprecated from components `manifest.json` file and should be moved into a `properties` dict.

For example:

```
-    {"type":"_airplay._tcp.local.","model":"appletv*"}
+    {"type":"_airplay._tcp.local.","properties":{"model":"appletv*"}}
```

Hedda, 2022-02-06:
By the way, Home Assistant founders/leads now explicitly said this year they will prioritize focus on features like these types:

https://www.home-assistant.io/blog/2022/01/19/streamlining-experiences/

Check out this video at around 11 min 55 sec in where talk about that focus point about making it easier to get started, etc.

https://www.youtube.com/watch?v=t_2D_KoFIfU&t=710s

Hopefully, it should be a greater chance of getting more help from other Home Assistant developers with this if ask them.

Hedda, 2022-05-10:
FYI, PR for ZiGate's Zeroconf has been merged in Home Assistant core making it an example for adding additional more readable:

https://github.com/home-assistant/core/pull/68577

https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/manifest.json#L85-L88

https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/config_flow.py#L198-L201

https://github.com/home-assistant/core/blob/dev/tests/components/zha/test_config_flow.py#L80-L107

https://github.com/home-assistant/core/blob/dev/homeassistant/generated/zeroconf.py#L384-L389

Hedda, 2025-01-20:
I believe this new feature in ZHA makes this old request obsolete? -> https://github.com/syssi/esphome-zb-gw03/issues/43

syssi, 2025-01-20:
Indeed! Superseeded by #43