syssi/esphome-zb-gw03 #7: How can pairing be started?

volkerrichert, 2021-12-16:


syssi, 2021-12-16:
As soon the gateway is added via the ZHA integration in Home Assistant there is a pairing button:

Settings -> Devices -> Search for "zha" -> Select the gateway device -> Click the button at the bottom of the device card.

[image]

syssi, 2021-12-16:
There is also a ZHA service called `zha.permit`. If you are keen you could connect the pairing button of the gateway to this service.

volkerrichert, 2021-12-16:
AH. ok. I try to user iobroker and the integrated esphome dashboard. The dashboard always said the gateway is offline. I'll take a look there

syssi, 2021-12-16:
https://esphome.io/guides/faq.html#docker-reference

> By default ESPHome uses mDNS to show online/offline state in the dashboard view. So for that feature to work you need to > enable host networking mode
>
> On MacOS the networking mode (“-net=host” option) doesn’t work as expected. You have to use another way to launch the > dashboard with a port mapping option and use alternative to mDNS to have the online/offline stat (see below)
>
> mDNS might not work if your Home Assistant server and your ESPHome nodes are on different subnets. If your router
> supports Avahi, you are able to get mDNS working over different subnets.
>
> Just follow the next steps:
>
> Enable Avahi on both subnets.
> 
> Enable UDP traffic from ESPHome node’s subnet to 224.0.0.251/32 on port 5353.
> 
> Alternatively, you can make esphome use ICMP pings to check the status of the device with the Home Assistant add-on
> "status_use_ping": true, option or with Docker -e ESPHOME_DASHBOARD_USE_PING=true. See also
> https://github.com/esphome/issues/issues/641#issuecomment-534156628.

It's your first ESPHome device, right?

volkerrichert, 2021-12-16:
Volltreffer ;-) Wollte dem Thema esphome einen Versuch geben, nachdem ich einiges gelesen / gehört hatte

syssi, 2021-12-16:
Does iobroker support EZSP radios attached via ser2net too? Could you provide some details? I would like to improve the documentation.

syssi, 2021-12-16:
https://forum.iobroker.net/topic/43446/zigbee-adapter-mit-sonoff-zigbee-bridge/22

volkerrichert, 2021-12-16:
i falshed a homebrow tasmota firmware with zb gateway support and add some scripts to handle the mqtt feedback inside iobroker. But not that handy oder elegant. And handling devices... i didn't touch that yet. A friend told me he is using esphome and i like the way of creating the firmware and an iobroker integration.

I found your repository and started with esphome. But i need HA to pair or is esphome dashboard enough? When i got my esp online ;-)

syssi, 2021-12-16:
Please try to understand this explanation first: https://github.com/syssi/esphome-zb-gw03/issues/2

Es ist wichtig, dass du zwischen den folgenden Lösungsansätzen unterscheidest: 

1. Standalone: Ich habe ein Funkmodul plus Mikrocontroller. Der Mikrocontroller (und häufig ein EEPROM als Storage) merken sich die Menge der verbundenen/gepaarten Geräte und in deinem Fall (Tasmota2Zigbee) publiziere ich den Status der Geräte samt aller Events über MQTT.

2. Modem: Ich bin ein Funkmodul plus Mikrocontroller und der Mikrocontroller wird lediglich verwendet, um die serielle Schnittstelle des Funkmoduls im Netzwerk verfügbar zu machen. Du kannst es dir auch als vorstellen, als haettest du dir einen Zigbee-USB-Stick gekauft. Anstatt mit diesem auf /dev/ttyUSB0 zu sprechen wird eine TCP-Verbindung zu einem Socket (IP+Port) aufgebaut. Hier fallen die seriellen Daten raus. Das Modem ist dumm. Weiss nichts und merkt sich nichts.

In Home Assistant gibt es eine Software-Implementierung von Zigbee namens ZHA. Hier konfiguriert man entweder einen USB-Stick oder einen Remote-Host + Port, hinter welchem ein Modem zu erreichen ist. Mir scheint es, als gaebe es in iobroker eine aehnliche Loesung. In diesem Fall managt die Softwareloesung die Menge der gepaarten Geräte. Deshalb darfst du dem Button zum pairen nicht hier im Projekt suchen, sondern musst die jeweilige Software-Implementierung befragen.

Du verwendest gerade Loesung 1. Dieses Projekt implementiert Loesung 2. Naemlich eine transparente Brücke zwischen Funkmodul und Software-Zigbee.

volkerrichert, 2021-12-16:
Danke für die ausführliche Antwort. Ich verstehe was du meinst und den Ansatz. Ich werde mal im Zigbee-Modul von ioBroker suchen. Ich habe im Kopf, dass da auch ein remote Support drin ist. Zur Not bau ich den ein.

Ich melde mich mit updates

Hedda, 2021-12-17:
> Does iobroker support EZSP radios attached via ser2net too?

FYI, ioBroker depends on zigbee-herdsman (which is the same Zigbee library that Zigbee2MQTT uses) and it does have initial experimental support for EZSP v8, but you should know that EZSP support is still in a very experimental phase in zigbee-herdsman o you should really post any EZSP specific feedback here -> https://github.com/Koenkk/zigbee-herdsman/issues/319 (though I do not know if there is currently any zigbee-herdsman developers actively working on improving or maintaining that initial EZSP support).

volkerrichert, 2021-12-17:
@Hedda good to know about the upcoming EZSP support.

closed 2022-02-22 by syssi