syssi/esphome-zb-gw03 #9: Low Transmission Power

alelucc, 2022-01-23:
While troubleshooting a problem i was facing, in the configuration.yaml i enabled ZHA logs and i noticed that the radio module was set with a very low Tx power (8).
`2021-01-25 09:04:00 INFO (MainThread) [bellows.zigbee.application] Node type: EmberNodeType.COORDINATOR, Network parameters: EmberNetworkParameters(extendedPanId=cc:cc:cc:cc:e3:ab:00:78, panId=0x3498, **radioTxPower=8,** radioChannel=25, joinMethod=<EmberJoinMethod.USE_MAC_ASSOCIATION: 0>, nwkManagerId=0x0000, nwkUpdateId=0, channels=<Channels.ALL_CHANNELS: 134215680>)`

And in some HA discussion i saw that some users having the same parameter set to 20. I think that is related to the Zigbee firmware that i flashed from [https://github.com/arendst/Tasmota/raw/development/tools/fw_SonoffZigbeeBridge_ezsp/ncp-uart-sw_6.7.8_115200.ota](url), is that correct? 
I was wondering if there were any "new" firmware with higher Tx power or maybe some way to change it myself.

Is here the rigth place to ask? Or maybe the github of the firmware developement?

syssi, 2022-01-23:
Let's discuss the question here too because it could be helpful to other users using the ESPHome approach.

I've checked the `bellows.zigbee.application` and it looks like the value is hardcoded at the initialization and passed to the device. Could you create an issue at the bellows repository and ask for a solution?

https://github.com/zigpy/bellows/blob/dev/bellows/zigbee/application.py#L230-L231

It looks like the `CONF_NWK_CHANNEL` can be set via the HA configuration already:

```
zha:
  zigpy_config:
    network:
      channel: 15             # What channel the radio should try to use.
      channels: [15, 20, 25]  # Channel mask
```

As similar solution for the TX power would be nice.

alelucc, 2022-01-23:
I have opened a new issue here [https://github.com/zigpy/bellows/issues/446#issue-1111893681](https://github.com/zigpy/bellows/issues/446#issue-1111893681)

Hedda, 2022-01-31:
Do not know for ezsp (bellows) but for znp (zigpy-znp) it uses firmware default unless set `tx_power` via ZHA YAML configuration:

https://github.com/zigpy/zigpy-znp/blob/dev/README.md#configuration

```
zha:
  zigpy_config:
    znp_config:
      # "auto" picks the largest value that keeps the device's transmit buffer from getting full
      max_concurrent_requests: auto

      # Only if your stick has a built-in power amplifier (i.e. CC1352P and CC2592)
      # If set, must be between:
      #  * CC1352/2652:  -22 and 19
      #  * CC253x:       -22 and 22
      tx_power:  
```

So if same command is available for ezsp in ZHA and in the bellow library then ZHA YAML configuration should be something like:

https://www.home-assistant.io/integrations/zha#configuration---yaml

```
zha:
  zigpy_config:
    ezsp_config:
      tx_power: 20
```

The default in firmware will probably be set to 10 dBm or less than that FCC and CE certification regulation does not allow higher.

Silicon Labs EFR32MG21 hardware does, however, support up to 20 dBm TX power but user need to set it themself if not default:

https://www.silabs.com/wireless/zigbee/efr32mg21-series-2-socs