syssi/esphome-zb-gw03 #11: [Feature Request] Compatibility with upcoming ITead Sonoff ZBBridge-P (ESP32 and CC2652P based)

Hedda, 2022-02-06:
This request comes a little ahead of its time since this upcoming ITead Sonoff ZBBridge-P is not available to buy yet, but will be soon ;)

https://community.home-assistant.io/t/itead-sonoff-zbbridge-p-coming-soon-a-new-cc2652-and-esp32-based-sonoff-zigbee-bridge-and-sensors-from-itead-leaked-via-fcc/378924

FCC leak shows the upcoming Sonoff ZBBridge-P v1.1 Zigbee Bridge coming from ITead will be based on ESP32 and CC2652P.

[https://fccid.io/2APN5ZBBRIDGEP](https://fccid.io/2APN5ZBBRIDGEP)

Internal pictures posted from FCC show using an unannounced Texas Intruments CC2652P (CC2652P1) MCU chip based “SM-031 v1.1” radio module which must be made by [CoolKit](https://github.com/CoolKit-Technologies/DevDocs/tree/master/Zigbee) (the company that makes their EFR32MG21 based “[SM-011 V1.0](https://www.coolkit.cn/product/sm-011/)” module). 

ESP32 has replaced the ESP8266/ESP8285, and it also looks like added an on-board clock-battery too, presumably for RTC (Real-Time Clock)?

https://fccid.io/2APN5ZBBRIDGEP/Internal-Photos/Internal-Photos-5613131

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

FFC also leaked new Sonoff branded [Door/Window Sensor](https://fccid.io/2APN5SNZB-04P), [Motion Sensor](https://fccid.io/2APN5SNZB-03P), [Temp/Humidity Sensor](https://fccid.io/2APN5SNZB-02P), and a [Button](https://fccid.io/2APN5SNZB-01P) as wireless Zigbee devices, all with the same CCC2652P (CC2652P1/CC2652P1F) chip which +20 dBm capable power amplifier is really overkill for battery-operated sensors and should on theory use more battery power than if they would have used the compatible CC2652R or CC2652RB.

Sadly it looks like even this new variant of Sonoff ZBBridge still uses Wi-Fi instead of wired Ethernet network like the newer ZB-GW03:

[https://community.home-assistant.io/t/zb-gw03-ewelink-ethernet-zigbee-gateway-now-hacked-with-tasmota-zbbridge-so-can-be-used-via-mqtt-or-as-a-remote-zigbee-adapter-with-home-assistant-zha-and-zigbee2mqtt/341223](https://community.home-assistant.io/t/zb-gw03-ewelink-ethernet-zigbee-gateway-now-hacked-with-tasmota-zbbridge-so-can-be-used-via-mqtt-or-as-a-remote-zigbee-adapter-with-home-assistant-zha-and-zigbee2mqtt/341223)

Ethernet is preferred by DIY users who flash Tasmota or ESPHome in order to use it as a remote Zigbee adapter with ZHA or Zigbee2MQTT because tunnelling serial communication over a serial stream server connection and the serial APIs of the Zigbee stacks are not robust/stable over WiFi as assume a direct connection, which is why both ZHA and Zigbee warns against using WiFi-based networked-attached Serial-to-IP Zigbee bridges/gateways/adapters:

[https://www.home-assistant.io/integrations/zha#warning-about-wi-fi-based-zigbee-to-serial-bridgesgateways](https://www.home-assistant.io/integrations/zha#warning-about-wi-fi-based-zigbee-to-serial-bridgesgateways)

[https://www.zigbee2mqtt.io/advanced/remote-adapter/connect_to_a_remote_adapter.html](https://www.zigbee2mqtt.io/advanced/remote-adapter/connect_to_a_remote_adapter.html)

According to digiblur who broke this news, these are slated to be released in March of 2022:

[https://www.digiblur.com/2021/12/new-sonoff-zigbee-bridge-and-sensors.html](https://www.digiblur.com/2021/12/new-sonoff-zigbee-bridge-and-sensors.html)

Not sure if this is intended to permanently replace the popular ITead Sonoff ZBBridge

[https://community.home-assistant.io/t/sonoff-zbbridge-sonoff-zigbee-bridge-from-itead/187346](https://community.home-assistant.io/t/sonoff-zbbridge-sonoff-zigbee-bridge-from-itead/187346)

As I understand ITead have not abandoned Silicon Labs yet but it is just that Silabs chip stocks have suffered worse than most other companies during the still currently ongoing worldwide chip shortage.

PS: Assume it will use the same firmware for CC2652P as “Sonoff Zigbee 3.0 USB Dongle Plus”:

[https://community.home-assistant.io/t/sonoff-zigbee-3-0-usb-dongle-plus-by-itead-is-based-on-texas-instruments-cc2652p-can-now-be-ordered-for-10-99/340705/283](https://community.home-assistant.io/t/sonoff-zigbee-3-0-usb-dongle-plus-by-itead-is-based-on-texas-instruments-cc2652p-can-now-be-ordered-for-10-99/340705/283)

Hedda, 2022-05-16:
FYI, both NotEnoughTech and digiblur have received theirs and while neither of them has tried to flash/hack it yet they posted some early thoughts on the original eWeLink hub firmware and hardware board, confirming that the first retail version ITead's SONOFF Zigbee Bridge Pro with PCB board labelled "ZBBridge-P V1.1 2021.11.12" is indeed based on Espressiff  ESP32-DOWD-V3 (that ESP32 variant by the way also has integrated support for Bluetooth 4.2 including BLE), Texas Instruments CC2652P Zigbee module labelled "SM-031 V1.1 2021.07.28" (so probably manufactured by [CoolKit Technologies](https://github.com/CoolKit-Technologies/DevDocs/tree/master/Zigbee) like the Zigbee module for previous Sonoff bridge), with a cell-battery for the RTC (Real-Time-Clock), a built-in buzzer for chime or alarm-siren, and also an Espressiff branded PSRAM (ESP-PSRAM16H) with 2MB/16Mbit capacity.

digiblurDIY also mention that he managed to backup original firmware from ESP32 without issue so suspect that it is not locked:

https://www.youtube.com/watch?v=P5op57y5i6k?t=920&ab_channel=digiblurDIY

https://notenoughtech.com/home-automation/sonoff-zigbee-bridge-pro/

https://www.youtube.com/watch?v=fsIjFxCPJLA&ab_channel=NotEnoughTECH

Hedda, 2022-12-27:
@syssi FYI, @jonasled created an ESPHome configuration for Sonoff ZBBridge-P bridge and posted it on GitLab here:

https://jonasled.dev/jonasled/sonoff-zigbee-bridge-pro-esphome

It is in turn based on the ESPHome configuration for the original Sonoff Zigbee Bridge from thegroove:

https://github.com/thegroove/esphome-zbbridge/issues/16

Hedda, 2023-02-10:
FYI, Giancky79 also created an ESPHome configuration for Sonoff ZBBridge-P bridge here:

https://github.com/Giancky79/ZB-Bridge-P

Giancky79, 2023-02-13:
hi, i don't know why, but i have the error from the log of zigbee2mqtt: "error failed to connect to the adapter (error srsp - sys - ping after 6000ms)", i found a new configuration for serial to try

serial:
  port: 'tcp://192.168.x.x:6638'
  adapter: zstack

it's normal? @Hedda i have update a fw with a python script, for change the string for working with cc2652 sm-031with latest update from fw of koenkk, i don't think the problem are on yaml esphome, i'm write correctly the fw of cc2652, but if this configuration working for zigbee2mqtt, update the github with all  
P.S. the python script is not mine but from @snakuzzo

Hedda, 2023-02-13:
Best is to post/ask under discussions in Koenkk's zigbee2mqtt repository -> https://github.com/Koenkk/zigbee2mqtt/discussions

Of course, you can not use `x` like `tcp://192.168.x.x:6638` as wildcards, instead, it needs to be the IP address of the device.

I also do not think you need to specify `adapter: zstack` as Zigbee2MQTT will default adapter to zstack if nothing is specified, so currently you only have to specify the specific radio type if using a ezsp, deconz, or zigate radio adapter and it is not detected.

https://www.zigbee2mqtt.io/advanced/remote-adapter/connect_to_a_remote_adapter.html

See -> https://www.zigbee2mqtt.io/advanced/remote-adapter/connect_to_a_remote_adapter.html#_3-configure

https://www.zigbee2mqtt.io/advanced/remote-adapter/connect_to_a_remote_sonoff_zbbridge.html#connect-to-a-remote-sonoff-zbbridge

PS: If to map the socket directly with socat (read example https://www.redhat.com/sysadmin/getting-started-socat ) then you can also for testing purposes connect to TCP port with zigpy-cli or/and zigpy-znp just to try see if applicaton firmware replies or not:

https://github.com/zigpy/zigpy-cli

https://github.com/zigpy/zigpy-znp 

You can for example test an energy scan with zigpy-znp tools -> https://github.com/zigpy/zigpy-znp/blob/dev/TOOLS.md

or do the same with zigpy-cli -> https://github.com/zigpy/zigpy-cli/blob/dev/README.md#performing-an-energy-scan

tube0013, 2023-02-13:
I have people encounter the "error srsp - sys - ping after 6000ms" randomly. the for a networ connected device, the most reliable way I've found to solve it is to do a full re-flash (including erase) of the esp device doing he serial stream. I still do not fully understand why this error occurs for some people and not others.

Giancky79, 2023-02-14:
> Best is to post/ask under discussions in Koenkk's zigbee2mqtt repository -> https://github.com/Koenkk/zigbee2mqtt/discussions
>
> Of course, you can not use `x` like `tcp://192.168.x.x:6638` as [...]

ops, i read an old documentation from zigbee2mqtt.io

[image]

Giancky79, 2023-02-14:
> I have people encounter the "error srsp - sys - ping after 6000ms" randomly. the for a network connected device, the most reliable way I've found to solve it is to do a full re-flash (including erase) of the esp device doing he serial stream. I still do not fully understand why this error occurs for some people and not others.

thx, i'm try with esptool, maybe esphome create a bin with unknown error inside or clone esp32, make different hardware problem

Giancky79, 2023-02-18:
ok, after few days, i'm try to format and re build esphome firmware, nothing change, after i'm try tasmota32-zbbrdgpro.factory, with new fw modified for zbbridge-p, zigbee2mqtt not working, same error srsp, with zha, after reconfigure tasmota with tcp, working very well.
I'm not an engineer, maybe it's my limit on compiling with esphome, but it didn't work, sorry for wasting everyone's time
i'm try with other configuration on esphome in the future

syssi, 2023-06-25:
Let's close this issue because there a other projects available already. I don't own the device and cannot help here much without having access to the hardware.

closed 2023-06-25 by syssi