syssi/esphome-zb-gw03 #12: [REQUEST] Zigbee Router configuration for ESPHome on ZB-GW03

Hedda, 2022-02-17:
FYI, xsp1989 has uploaded a signed Zigbee Router firmware for ITead's Sonoff ZBBridge which could possibly make it a Tasmota/ESPHome connected Zigbee Router (instead of as a Zigbee Coordinator).

I think that this firmware image should also be compatible with ZB-GW03 as it too uses "SM-011 V1.0"? 

Do you think that ESPHome could send basic commands to this Zigbee Router firmware to initiate paring/joining and restart/reset?

> [https://drive.google.com/file/d/1H-M5CTh_XGGVl2te4vHLQ5SfGPHL6yCS/view?usp=sharing](https://drive.google.com/file/d/1H-M5CTh_XGGVl2te4vHLQ5SfGPHL6yCS/view?usp=sharing)
> 
> This is the signed routing firmware used by ZBB, the usage method is the same as the unsigned firmware.
> 
> [https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle/RouterForDongle/README.md](https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle/RouterForDongle/README.md)

Idea around this is also discussed for ITead's Sonoff ZBBridge hacked with Tasmota/ESPHome here:

[https://github.com/arendst/Tasmota/discussions/11536](https://github.com/arendst/Tasmota/discussions/11536)

[https://github.com/thegroove/esphome-zbbridge/issues/3](https://github.com/thegroove/esphome-zbbridge/issues/3)

[https://github.com/xsp1989/zigbeeFirmware/issues/2](https://github.com/xsp1989/zigbeeFirmware/issues/2)

[https://github.com/xsp1989/zigbeeFirmware/issues/16](https://github.com/xsp1989/zigbeeFirmware/issues/16)

I have not tried this myself as I got rid of my ITead Sonoff ZBBridge because being WiFi-based it did not work any good as a remote Zigbee Coordinator, and never bought a ZB-GW03, but might reconsider buying these if it worked as a Zigbee Router device that can be paired/joined and restarted/reset remotely via Tasmota and/or ESPHome. So that Tasmota/ESPHome is basically only used to initiate virtual join/pair button and restart/reset the device similar to a remote power-cycle it hangs.

Hedda, 2022-02-18:
Update from xsp1989 on router firmware posted on https://github.com/arendst/Tasmota/discussions/11536

"This firmware can ground the PA00 5S into pairing mode, and ground the RST to restart routing. UART is used to transmit data transparently.

and

"There is also another version of the routing firmware that can control the join network through a proprietary serial port protocol"

Hedda, 2022-02-20:
Check out this new "How to Convert the Sonoff Zigbee Bridge into a Router / Repeater" blog article by @digiblur here:

https://www.digiblur.com/2022/02/convert-sonoff-zigbee-bridge-to-router-repeater.html

Note that the article already assumes that you have flashed Tasmota onto ITead Sonoff ZBBridge as per his previous guide:

https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html

syssi, 2022-02-22:
```
{
  "NAME":"Sonoff ZigRouter","GPIO":
  [
    320,   # GPIO0     Led_i 1     Inverted LED - default state ON
    5088,  # GPIO1     TCP TX      TCP to serial bridge
    0,     # GPIO2     Not used
    5120,  # GPIO3     TCP RX      TCP to serial bridge
    257,   # GPIO4     Relay_i 2   Relay inverted
    256,   # GPIO5     Relay_i 1   Relay inverted
    0,     # GPIO6     Not used
    0,     # GPIO7     Not used
    0,     # GPIO8     Not used
    576,   # GPIO9     LedLinki    Device Status LED, inverted
    0,     # GPIO10    Not used
    0,     # GPIO11    Not used
    32,    # GPIO12    Button 1    Button active low, internal pull-up resistor
    0      # GPIO13    Not used
   ],
  "FLAG":0,
  "BASE":18
} ; delay1 ; module 0
```

Summary: `GPIO5` is used to trigger to pairing mode (via Relay_i  1 or Button 1) at digiblurs tasmota template.

syssi, 2022-02-22:
Comparison of different Sonoff Zigbee Bridge tasmota templates:

```
{"NAME":"ZHA ZBBridge",    "GPIO":[ 56, 208,0, 209, 59, 58,0,0,0,  0,0,0,17  ],"FLAG":0,"BASE":18}
{"NAME":"Sonoff ZHABridge","GPIO":[ 56, 208,0, 209, 59, 58,0,0,0,  0,0,0,17  ],"FLAG":0,"BASE":18} ; module 0
{"NAME":"Sonoff ZigRouter","GPIO":[320,5088,0,5120,257,256,0,0,0,576,0,0,32,0],"FLAG":0,"BASE":18} ; delay1 ; module 0 
```

syssi, 2022-02-22:
https://github.com/xsp1989/zigbeeFirmware/issues/16#issuecomment-1044625713
https://github.com/arendst/Tasmota/discussions/11536#discussioncomment-2202272

```
"This firmware can ground the PA00 5S into pairing mode, and ground the RST to restart routing. UART is used to transmit data transparently.
```

At the ZB-GW03 `PA00` is attached to `GPIO12` of the ESP: https://github.com/syssi/esphome-zb-gw03/blob/main/example.yaml#L93-L99

syssi, 2022-02-22:
I've added a guide to use the ZB-GW03 as Zigbee router: https://github.com/syssi/esphome-zb-gw03/blob/main/docs/router.md

digiblur, 2022-02-22:
I kept it simple with the pairing and zigbee reset for folks.

Hedda, 2022-02-23:
@syssi Could you copy the button press paring mode option from @digiblur as well as too is a great alternative option?

https://www.digiblur.com/2022/02/convert-sonoff-zigbee-bridge-to-router-repeater.html

_Next we need to enable the ability for the GUI and reset button to be used as the device pairing procedure.  Paste the following command into the console all on one line:_

`backlog template {"NAME":"Sonoff ZigRouter","GPIO":[320,5088,0,5120,257,256,0,0,0,576,0,0,32,0],"FLAG":0,"BASE":18} ; delay1 ; module 0`

The bridge should reboot and come back up on the console.  The device is ready to be paired with your Zigbee network.  Go _ahead and place it in the desired location of the home and power it up.  Enable pairing on your Zigbee Network as you would normally pair a regular sensor, switch, etc.  NOTE: Zigbee2MQTT users as of this time will need to add a custom file to support this router before pairing (see last section of this guide)_

_To place the device into pairing mode, simply use a paper clip to briefly push the reset button next to the USB power plug._ 

[image]

_The green LED will light up.  Allow 5-6 seconds to pass and press the button again, this will turn off the green LED and the device will be in pairing mode._

[image]

syssi, 2022-02-23:
This is already the case: https://github.com/syssi/esphome-zb-gw03/blob/main/packages/button_pairing_mode.yaml#L21-L31

If you press the button of the device the pairing mode is triggered. On difference: There is no LED attached.

commit 42f30607 (2022-02-22): Add router support (Closes: #12) (#14)