syssi/esphome-zb-gw03 #18: Am I missing something?

brandonb1987, 2022-09-13:
Was able to get the code onto the device up to the point where I need to download the zigbee firmware.

I'm not sure what the instructions refer to with all the code below the "download and reset button" but when i do that step, nothing happens.

If i hit the download it gives me this in the log
[22:58:39][D][switch:013]: 'zb-gw03 Zigbee Download Mode' Turning ON.
[22:58:39][D][switch:037]: 'zb-gw03 Zigbee Download Mode': Sending state ON

If i hit the reset after I get this....no zigbee radio firmware downloads?
[22:59:56][D][switch:017]: 'zb-gw03 Zigbee nRST' Turning OFF.
[22:59:56][D][switch:037]: 'zb-gw03 Zigbee nRST': Sending state OFF

I also don't ever get the temperature

Any help?

syssi, 2022-09-13:
You are trying to convert your device to a Zigbee router and you got stuck [here](https://github.com/syssi/esphome-zb-gw03/blob/main/docs/flashing.md#flash-the-zigbee-module), right?

syssi, 2022-09-15:
@brandonb1987 Friendly ping.

syssi, 2022-09-19:
Some background: The ZB-GW03 is equipped with a Zigbee module. If we want to update the firmware of the zigbee module we must enable the "download mode" and reboot the Zigbee module (using the `zigbee reset` switch). The module boots into the "download mode" now.

We can talk to the serial interface of the Zigbee module using telnet (a network connection) because the ESPHome node provides a little ser2net service here. The code block of the documentation shows a telnet session. If you telnet to your ESPHome node on port 6638 and press return the Gecko bootloader of the zigbee module should respond with:

```
Gecko Bootloader v1.9.1.04
1. upload gbl
2. run
3. ebl info
BL >
```

This is a "menu". If you type `1` and press return the Bootloader starts into the "upload gbl" mode and asks you to upload the new firmware. Please abort the telnet session now because we don't use the telnet client to upload the binary. We use another tool/client called `sx`: 

```
sx -vv -X -b --tcp-client 192.168.132.225:6638 ncp-uart-sw_6.7.8_115200.ota
```

This command establishes a new telnet connection to the ESPHome node and uploads the OTA file to the Zigbee module. You should see some progress and a `Transfer complete` response.

As last step we want to boot into the new firmware and don't want to stop at the Gecko bootloader again. You have to turn off the "download mode" switch and restart the Zigbee module using the "zigbee reset" switch.

brandonb1987, 2022-09-26:
It's strange because I did not have to do any of that in the end. It just started working all of a sudden. Not sure if that is because I went from beforehand Tasmota where I used the same .ota firmware or not, but it works.  

The write-up you just added makes WAY more sense now. I suggest you add the extras to the instructions in the readme as it will make people stop scratching their head around that point lol. I didn't realize everything was done over telnet in that phase. The instructions at that point weren't very clear to the 'non-tech savy'.

Thanks! 

The only concern I have is it pulls a bunch of stuff from different sources. This is quite reliant on git, and their maintainers continuing the good work.  Any intentions to make it not pull from github?

syssi, 2022-09-26:
Your guess about the Zigbee firmware is correct. If you have used tasmota beforehand you had flashed the alternative Zigbee firmware already.

You could make a local copy of all external components and change the `external_components` section to pull from the local filesystem.

cptbucky, 2023-03-18:
First off, thank you ever so much for all your work here, I now have this running with Zigbee2MQTT which has been the plan for a very long time!

This issue explains a lot so thank you for logging, as I hit the telnet section and due to lack of experience I didn't know what to do, I did hit enter however I didn't get the gecko bootloader so skipped it and everything just worked. It appears that I'm in the same boat as @brandonb1987 as I had also previously converted the device to Tasmota so got lucky!

It might be worth adding a brief on Zigbee2MQTT as there is so little out there that this repo is absolute gold but hard to find when not looking for ZHA, though obviously not impossible :-)

syssi, 2023-03-19:
@cptbucky Could you describe your setup in a few words? I must admit I don't know zigbee2mqtt in detail and I don't feel confident to add details / instructions to the docs.

cptbucky, 2023-03-19:
I currently run a docker host which has Home Assistant, ESPHome, Zigbee2MQTT and Mosquitto Broker to name the important bits. Zigbee devices are predominantly attached to a ConBee II that is attached to the server and mounted to Zigbee2MQTT, though I've now added the GW-03 also, with a 2nd Zigbee2MQTT instance, same broker.

The "Add the ESPHome node to Home Assistant" section in the docs is slightly different for me as I don't use the HASS integration and haven't fixed an mDNS issue that sees me chasing static IPs :pleading_face: 

The "Connect the device to the ZHA integration of Home Assistant" section in the docs has all the information except that for Zigbee2MQTT you can either go to the application itself and setup the adapter as follows:

[image]

I will have configured via the config files however, which is an entry of the following:

```
serial:
  port: tcp://192.168.1.100:6638
  adapter: ezsp
```

Official docs can be found here: https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html

If you like I can take a stab at a PR to list out both ZHA and Zigbee2MQTT options?

syssi, 2023-03-19:
I would super happy about a PR! Let's add a new section to the `flashing guide` mentioning the zigbee2mqtt route.

Bytchus, 2023-05-19:
> Please abort the telnet session now because we don't use the telnet client to upload the binary. We use another tool/client called `sx`:

Good day, 
First of all, say that I am impressed with all the work here selflessly shared, it is great, thank you very much.

I will try to follow your flashing guide from Windows, is there any client that I could use to upload the file from Windows? 

Thanks.

syssi, 2023-05-22:
You can use any terminal program which supports network connections & xmodem data transfer. It looks like ExtraPuTTY could do the job!

Bytchus, 2023-05-26:
Thx, I will follow it.

ivananasco, 2024-04-04:
i trying install firmware zigbee....

ubuntu@IvanPC:~$ telnet 192.168.1.23 6638
Trying 192.168.1.23...
Connected to 192.168.1.23.
Escape character is '^]'.
1

no display Gecko Bootloader v1.9.1.04.....