syssi/esphome-zb-gw03 #26: Cannot update with ESPhome after 2023.2.3 OTA

maegibbons, 2023-04-06:
Hi

ESPhome updated to 2023.3.2 recently so I tried to update my gateway from 2023.2.3 firmware to 2023.3.2 firmware wirelessly.

Compiled and links fine
I get Error 104 when it tries to upload

```
INFO Reading configuration /config/esphome/zb-gw03-coordinator.yaml...
INFO Updating https://github.com/syssi/esphome-zb-gw03@main
INFO Reverting changes to https://github.com/syssi/esphome-zb-gw03@main -> 06937b87edf59b7afa84f06d3dffa870b38335ba
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO2 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO12 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO15 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing zb-gw03-coordinator (board: esp-wrover-kit; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
|-- Wire @ 2.0.0
Compiling /data/zb-gw03-coordinator/.pioenvs/zb-gw03-coordinator/src/main.cpp.o
Linking /data/zb-gw03-coordinator/.pioenvs/zb-gw03-coordinator/firmware.elf
RAM:   [=         ]  10.7% (used 34944 bytes from 327680 bytes)
Flash: [===       ]  32.6% (used 597981 bytes from 1835008 bytes)
Building /data/zb-gw03-coordinator/.pioenvs/zb-gw03-coordinator/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin(["/data/zb-gw03-coordinator/.pioenvs/zb-gw03-coordinator/firmware.bin"], ["/data/zb-gw03-coordinator/.pioenvs/zb-gw03-coordinator/firmware.elf"])
Wrote 0xa2510 bytes to file /data/zb-gw03-coordinator/.pioenvs/zb-gw03-coordinator/firmware-factory.bin, ready to flash to offset 0x0
========================= [SUCCESS] Took 9.24 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of zb-gw03-coordinator.local
INFO  -> 192.168.29.47
INFO Uploading /data/zb-gw03-coordinator/.pioenvs/zb-gw03-coordinator/firmware.bin (599312 bytes)
Uploading: [=                                                           ] 2% 
ERROR Error sending data: [Errno 104] Connection reset by peer

```

How do I overcome this?

Krs

Mark

syssi, 2023-04-06:
Ping me if you have trouble to reflash the device but I guess you will manage it like the first time. Another idea: If you reboot the device 10 times in a row ESPHome should go into safe mode. If you are lucky OTA will succeed here because of the reduced load.

syssi, 2023-06-25:
I've updated my device to `2023.6.2` today via OTA without issues.

closed 2023-06-25 by syssi