syssi/esphome-zb-gw03 #30: ZB-GW03-V1.4

Slonik, 2023-09-15:
Is the current firmware suitable for version 1.4?
[image]

syssi, 2023-11-07:
Could you provide a photo of the upper side of the board. I would like to see / identify the different chips.

syssi, 2023-11-07:
https://github.com/arendst/Tasmota/discussions/12764#discussioncomment-6340566

```
backlog template {"NAME":"ZHA-bridge","GPIO":[0,0,5472,0,5504,0,0,0,5793,5792,320,544,5536,0,5600,0,0,0,0,5568,0,0,0,0,0,0,0,0,608,640,32,0,0,0,0,0],"FLAG":0,"BASE":1} ; module 0
``` 

In general the hardware version should be suitable but it's untested and unverified yet.

gdanov, 2024-01-06:
I have that version. Flashed esphome without any issues, ZB fw upload works, everything's great.

closed 2024-01-07 by syssi

lcavalli, 2025-09-06:
Confirmed v1.4 working also as zigbee proxy using provided instructions.

commit 84629a39 (2024-01-07): Add ZB-GW03-V1.4 as supported (Closes: #30) (#32)