syssi/esphome-zb-gw03 #31: no ACK at the end of flashing

gdanov, 2024-01-06:
converted my gateway from tasmota (stock 13.2, self upgraded to latest 13.3) and esphome works great.
then followed the zigbee flashing section, trying to flash fresher firmware, but regardless if I use more recent or the approved version, my upload looks like this 
```
$ sx -vv -X -b --tcp-client 192.168.1.72:6638 ncp-uart-sw_7.3.1.0_115200.gbl
connecting to [192.168.1.72] <6638>

Sending ncp-uart-sw_7.3.1.0_115200.gbl, 1819 blocks: Give your local XMODEM receive command now.
Xmodem sectors/kbytes sent: 1819/227kRetry 0: No ACK on EOT

Transfer incomplete
```
the hub works and zb2mqtt reports the stock firmware on the settings/about page

so, few questions
* is the transfer really incomplete? if so, what am I doing wrong?
* is the settings/about firmware version reliable?

gdanov, 2024-01-06:
Hm, if I use the `.ota` version of the firmware the finale is as expected. Obviously the format matters but I don't know how to get `.ota` for the latest fw version.

syssi, 2024-01-06:
I don't know the difference between GBL and OTA files. I'm not very deep into the topic. If you still want to flash this recent firmware version you could solder some wires into the device again and flash the zigbee modem directly using a USB-to-TTL adapter:

https://github.com/NabuCasa/universal-silabs-flasher?tab=readme-ov-file#flashing-firmware

gdanov, 2024-01-07:
> I don't know the difference between GBL and OTA files. I'm not very deep into the topic. If you still want to flash this recent firmware version you could solder some wires into the device again and [...]

thanks, here is the authoritative explanation https://github.com/xsp1989/zigbeeFirmware/issues/48#issuecomment-1879895599

I guess I'll sit tight for the time being :) it works great anyway.

syssi, 2024-01-07:
Perfect! Thanks for your feedback. Let's close this issue.

closed 2024-01-07 by syssi