syssi/esphome-zb-gw03 #34: Add WiFi support

OToussaint, 2024-01-10:
I succesfully flashed the router-example to the GW (1.3).
But I'm using it exclusivelly over WiFi.
And only ethernet is working.
Trying to add a wifi section in the yaml file is generating an error:

`
ethernet: [source <unicode string>:38]
  
  Component ethernet cannot be used together with component wifi.
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 1
  power_pin: GPIO16
`

syssi, 2024-01-10:
Please remove the `ethernet` section. You cannot use both simultaneously.

OToussaint, 2024-01-10:
Obviously I need to.
But the full original file is
```
substitutions:
  name: zb-gw03-router
  device_description: "ZB-GW03 eWeLink Ethernet Zigbee Gateway based Zigbee Router"

packages:
  zb-gw03:
    url: https://github.com/syssi/esphome-zb-gw03
    ref: main
    files:
      - packages/core.yaml
      - packages/status_led.yaml
      - packages/button_pairing_mode.yaml
    refresh: 0s
```
No ethernet section.

Do I need to fork the repo to do my own 
`      - packages/core.yaml`

And the original GW firmware was able to manage both, that would be awesome if the ESPHome version could do the same.

syssi, 2024-01-10:
Good point. I will try to split the core package to make the usage of the `ethernet` component optional.

OToussaint, 2024-01-10:
Thank you :)
I see that it is a limitation of ESPHome to prevent usage of Ethernet and WiFi at the same time, I'm sad.
Btw, do you know if it is possible to make the green led available (and always on as soon as it is connected to the network)?

syssi, 2024-01-10:
I will double check `GPIO14` can be controlled or not. I guess it can be exposed as `light` entity.

syssi, 2024-01-10:
At the moment the red LED indicates critical errors using the `status_led` component. A missing network connection is a critical error as far as I know.

syssi, 2024-01-10:
I've prepared a new release to make the `ethernet` component optional. Please give it a try!

syssi, 2024-01-10:
GPIO14 is exposed as monochromatic light now. A PWM signal is used to control the brightness if you like.

OToussaint, 2024-01-10:
Almost :) ethernet section is still in core.yaml

syssi, 2024-01-10:
Oh, noo. :-) I will push fix

syssi, 2024-01-10:
Finally. Please give it another try! :-)

closed 2024-01-10 by OToussaint