syssi/esphome-zb-gw03 #39: No longer able to flash updated ESPHome versions: Error: Authentication invalid. Is the password correct?

louispires, 2024-05-22:
I flashed this well over a year ago, worked great until about August last year when the ESPHome firmware no longer flashed successfully via OTA.

```
Building .pioenvs/zigbee/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin([".pioenvs/zigbee/firmware.bin"], [".pioenvs/zigbee/firmware.elf"])
Wrote 0xdc2d0 bytes to file /data/build/zigbee/.pioenvs/zigbee/firmware-factory.bin, ready to flash to offset 0x0
======================== [SUCCESS] Took 380.16 seconds ========================
INFO Successfully compiled program.
INFO Connecting to 10.0.0.31
INFO Uploading /data/build/zigbee/.pioenvs/zigbee/firmware.bin (836304 bytes)
ERROR Error auth result: Error: Authentication invalid. Is the password correct?
```

I tried playing around with the OTA Password field etc. but nothing seems to have worked and I am now stuck on an old version

```
substitutions:
  name: zigbee
  device_description: "ZB-GW03 eWeLink Ethernet Zigbee Gateway based Zigbee Coordinator"

packages:
  zb-gw03:
    url: https://github.com/syssi/esphome-zb-gw03
    ref: main
    files:
      - packages/core.yaml
      - packages/status_led.yaml
      - packages/green_led.yaml
      - packages/i2c.yaml
      - packages/button_zigbee_reset.yaml
      - packages/mdns.yaml
    refresh: 0s

# Enable Home Assistant API
api:
  encryption:
    key: "bQ30ENXTK+FZ+8X5ryGB0HTzoulLftOS8fiyHzMUbpg="

ota:
  password: "a6c199f4e53f394834c47b3b5e0bbedb"

# Enable logging
logger:

wifi:
  ssid: !secret wifi_ssid_iot
  password: !secret wifi_password_iot

  manual_ip:
    static_ip: 10.0.0.31
    gateway: 10.0.0.1
    subnet: 255.255.255.0
```

Will probably need to sort this out soon due to the [deprecation of the ezsp](https://github.com/Koenkk/zigbee2mqtt/discussions/21462) and seems like [xsp1989](https://github.com/xsp1989) is busy attempting a new [firmware build](https://github.com/xsp1989/zigbeeFirmware/issues/49)

Ideally, I would not like to resolder again and prefer to be able to do an OTA of some sort.

Any recommendations?

syssi, 2024-05-22:
You are on the right track. You have to find out the OTA password or key you have used last time to flash the device. Did you migrate to the encryption key already or did you flash the device using an OTA password last time? If you are unable to find out the password you cannot flash the device using OTA. Soldering would be the only option I'm aware of.

louispires, 2024-05-22:
> You are on the right track. You have to find out the OTA password or key you have used last time to flash the device. Did you migrate to the encryption key already or did you flash the device using an OTA password last time? If you are unable to find out the password you cannot flash the device using OTA. Soldering to flash the ESP using an UART connection would be the only option I'm aware of.

I had never added an OTA password, it just randomly one day stated that the password was incorrect.

That is when I started experimenting with adding a value and even attempting to set the password on boot:
```
on_boot:
    - lambda: |-
        id(my_ota).set_auth_password("New password");
```

Which also did not work for me...

syssi, 2024-05-24:
Do you agree to close the issue?

closed 2024-08-12 by syssi