syssi/esphome-zb-gw03 #40: Flashing from Tasmota

mrbrdo, 2024-06-04:
I have a zb-gw03 that came already flashed with Tasmota 13.2.0 (from Aliexpress). Can I flash esphome directly from the Tasmota firmware upgrade? If yes, perhaps it would be nice to add some short instructions for this case? As far as I read here, it seems flashing the zigbee module is also not necessary in this case.

Also if I can ask here - it seems most files are about 2 years old, were there no updates necessary? And are there any plans regarding the deprecation of ezsp?

Thanks

syssi, 2024-06-04:
This is the official migration guide how to go from tasmota to ESPHome without soldering: 
https://esphome.io/guides/migrate_sonoff_tasmota.html

Most of the time a update of the zigbee firmware isn't needed but it depends of the already installed version.

There are no plans about the depreciation of ezsp yet.

syssi, 2024-06-06:
Hmm.. I wasn't aware of this detail. That's bad. In this case you have to open the device and solder some wires the UART interface of the ESP.

syssi, 2024-06-06:
Do you agree to close this issue?

mrbrdo, 2024-06-06:
Yes, thanks for your help!

closed 2024-06-06 by mrbrdo