syssi/esphome-zb-gw03 #42: Z2M Deprecated driver 'ezsp' currently in use

a-d-r-i-a-n-d, 2024-08-12:
Hi @syssi,

First, thanks for putting everything together to make things easier for newbies like myself. I've noticed a warning in the Zigbee2MQTT logs that looks like this:
```
[2024-08-12 02:40:24] warning: 	zh:ezsp: Deprecated driver 'ezsp' currently in use, 'ember' will become the officially supported EmberZNet driver in next release. If using Zigbee2MQTT see https://github.com/Koenkk/zigbee2mqtt/discussions/21462
```
I read through that discussion, but I couldn't find any non-serial Ember firmware for the zb-gw03. It seems they mostly tested the serial Zigbee adapters. Is it possible to switch from `ezsp` to `ember` on the zb-gw03? Would the flashing process remain the same, using telnet and lsx, but uploading a `.gbl` file instead of a `.ota`? Any help is much appreciated.

syssi, 2024-08-12:
It looks like updating the zigbee radio firmware to https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/ZigbeeBridge_SM-011-signed/ncp-uart-sw_7.4.3.0_115200.ota and changing the driver from `ezsp` to `ember` at zigbee2mqtt should do the job.

https://github.com/arendst/Tasmota/discussions/21361#discussioncomment-9345723
https://github.com/xsp1989/zigbeeFirmware/issues/49#issuecomment-2137189283

a-d-r-i-a-n-d, 2024-08-12:
Thanks, I was already on `ncp-uart-sw_7.4.3.0_115200.ota` and yes I can confirm that switching the driver from `ezsp` to `ember` did the job.

closed 2024-08-12 by a-d-r-i-a-n-d