syssi/esphome-zb-gw03 #43: Support the new/upcoming generic ZHA Zeroconf discovery?

Hedda, 2024-09-30:
puddly is working on a new new/upcoming generic Zeroconf discovery for the ZHA integration in Home Assistant to allow it to discover new network-attached Zigbee Coordinator adapters without code changes to the ZHA component in Home Assistant core. Please look into adding support for this Zeroconf discovery formatting, see:

https://github.com/home-assistant/core/pull/126294

The goal with this is that code changes to the ZHA component in Home Assistant core but it puts some new requirement on the firmware site where the manufacturer need to enter unique metadate for each new type and model of network-attached Zigbee Coordinator adapter. This will eventeully also make the old disovery method obeslete:

https://www.home-assistant.io/integrations/zha#discovery-via-usb-or-zeroconf

PS: Alternative is to use the old method but it is likley that new patches to Home Assistant core will be accepted once support for this new formallting/method has been implemented.

Hedda, 2025-01-20:
@syssi Bump since this "generic ZHA Zeroconf discovery" feature has been merged into Home Assistant's core 

* https://github.com/home-assistant/core/pull/126294

"_*I'm introducing a new service type: `_zigbee-coordinator._tcp.local.`. This will allow all future TCP coordinators to be supported without an explicit PR to ZHA. The only expected properties are `radio_type` (one of `ezsp`, `znp`, etc.) and `serial_number` (must be unique per device and not derived from any Zigbee information, like the Ethernet MAC address). The service name will be shown as-is*_."

syssi, 2025-01-20:
I will give it a try soon:

```diff
diff --git a/packages/mdns.yaml b/packages/mdns.yaml
index ce8ab52..8e731c2 100644
--- a/packages/mdns.yaml
+++ b/packages/mdns.yaml
@@ -1,9 +1,10 @@
 mdns:
   services:
-    - service: _tube_zb_gw_efr32
+    - service: _zigbee-coordinator
       protocol: _tcp
       port: 6638
       txt:
+        # serial_number: aabbccddeeff
         name: ZB-GW03
         version: 1.0
         radio_type: ezsp
```

Hedda, 2025-01-20:
Tip is to check out code from [tube0013](https://github.com/tube0013) since read that he helped puddly beta test it so also first to implement this feature:

* https://github.com/tube0013/TubesZB-ESPHome-Builder

Noted that he is using get_mac_address and convert that into a serial_number in order to make the serials unique for each device:

* https://github.com/tube0013/TubesZB-ESPHome-Builder/blob/main/esphome-config/tubeszb-efr32-mgm210-poe-2023.yaml#L10

PS: Off-topic but you might also be interested to check out his customized version of the [uart](https://esphome.io/components/uart.html) component which enable hardware flow control for ESP32 boards:

* https://github.com/tube0013/esphome-components

syssi, 2025-01-20:
This is an awesome solution! I wasn't a fan of the hardcoded serial number of my draft.

syssi, 2025-01-20:
@Hedda Could you create another issue for hardware flow control? To discuss the topic. I guess it requires a hardware modification.

syssi, 2025-01-22:
I'ts done and it's working great. I appreciate your determination!

closed 2025-01-22 by syssi