syssi/esphome-zb-gw03 #45: No ACK on EOT

medassi, 2024-12-21:
Hi,
After many attempts, I still encounter this error:
Xmodem sectors/kbytes sent: 1780/222k
Retry 0: No ACK on EOT
The upload fails before completion. I've tried with all the firmware versions, and the result is always the same.

I've updated the ESPHome layer to 2024.12.2, and HA recognizes it as ESPHome 2024.12.2, so that part seems fine.
The current Zigbee firmware is 6.7.10.0 build 423.
I need to update my ZB-GW03-1.4 gateway to make it Ember-compatible with HA.

What am I missing?

syssi, 2024-12-21:
Could you tell me the exact filename and source of your zigbee firmware file? Why do you prefer version 6.7 over 7.4?

medassi, 2024-12-22:
I don’t prefer version 6.7. I want version 7.4, but I have tested the others as well. The firmware file is from xsp1989.

syssi, 2024-12-22:
Do you like to provide some more details? Which file did you use? I can offer to write the mentioned firmware to my device just for testing.

medassi, 2024-12-23:
Ok thanks a lot. Here the firmware file : https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/ZigbeeBridge_SM-011-signed/ncp-uart-sw_7.4.3.0_115200.ota
I hope it's correct.

syssi, 2024-12-23:
1. Turn on the "download mode" switch
2. Turn on the "zigbee reset" switch
3. Open a terminal an telnet into the bootloader and type `1<return>` to select the "upload gbl" option

[image]

4. Don't close the telnet session. Go to a second terminal and start the upload:

[image]

It's working fine on my device.

medassi, 2024-12-23:
Ok thank you for this test. That doesn't reassure me. I do exactly the same thing but it doesn't happen the same. the upload runs correctly up to 222k and then stops, informing me that the upload is incomplete. I don't see where the error comes from at all. My gateway comes from Ali express and is stamped zb-gw-03 v1.4. On the other hand, it is not noted SM-011 on the chipset. Thanks anyway, I'll keep looking.

medassi, 2024-12-23:
[image]

syssi, 2024-12-23:
May be the Zigbee module rejects the new firmware for unknown reason. Which debian / lrzsz version do you use?

```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm
$ dpkg -l | grep lrzsz
ii  lrzsz                                                       0.12.21-10+b1                                                              amd64        Tools for zmodem/xmodem/ymodem file transfer
$
```

medassi, 2024-12-24:
[image]
It seems that we have the same versions though.

syssi, 2024-12-24:
Let's compare the versions of the Gecko Bootloader as next step. Could you telnet into your bootloader another time and have a look?

medassi, 2024-12-25:
Yes of course, but it's the same as your :
[image]

medassi, 2024-12-25:
oops !!! I'm really, really sorry for wasting your time. Everything works. I was using a simple wget to recover the ota file and I had to recover a corrupted file. I just redownloaded it directly from the github repository and sent it with winscp to my server. And there, the sx worked. Once again, sorry and many thanks for your perseverance. The bug is still between the chair and the keyboard...
[image]

medassi, 2024-12-25:
All seems to be ok 👍 
[image]

syssi, 2024-12-25:
Awesome! No problem. I'm happy you was able to resolve the issue. Merry christmas!

closed 2024-12-25 by syssi

syssi, 2025-03-03:
> wget https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/ZigbeeBridge_SM-011-signed/ncp-uart-sw_7.4.3.0_115200.ota

This doesn't work. Please check the content of the `ncp-uart-sw_7.4.3.0_115200.ota` file. I guess it's HTML.

This command should download the binary file:

```
wget https://github.com/xsp1989/zigbeeFirmware/raw/refs/heads/master/firmware/ZigbeeBridge_SM-011-signed/ncp-uart-sw_7.4.3.0_115200.ota
```

ElVit, 2025-03-03:
Thank you, this was the problem. Now its working 😃