syssi/esphome-zb-gw03 #61: I2C?

ElVit, 2025-07-05:
Hi,
I am using also your code and it works perfect. 
But I have wondered why there is an I2C component in your yaml code. 
There is no other component in the yaml using this I2C connection. 
If I see the logs there have been indeed some I2C addresses found:
```
[21:47:57][I][i2c.idf:104]: Results from bus scan:
[21:47:57][I][i2c.idf:110]: Found device at address 0x50
[21:47:57][I][i2c.idf:110]: Found device at address 0x58
```
Do you know to what module this addresses belong?
Can we do something with this I2C module?

syssi, 2025-07-05:
I've added i2c because I would like to show there is a bus with some devices attached to it. You could attach additional devices (supported by ESPHome) if you like. I did not check what's 0x50 and 0x58 in detail.

closed 2025-07-06 by syssi

ElVit, 2025-07-06:
Thanks for your answer. 
I found the following ICs on my PCB:
- winbond 25Q32JVSIQ 2422
—> 3V 32M-Bit serial flash memory with dual, quad SPI
- SMSC 8720A 221 AUTA BTW
—> Ethernet Transeiver Module
- ESP32 V3 302024 HGOOMBG007
—> ESP32 Module
- FM24C512D 2426JDG1
—> 2-Wire Serial EEPROM
- Pulse H1102NL 2443-CHINA
—> LAN Discrete Transformer Module
- CJT1117B 3.3 WG
—> Voltage Regulator (probably 5V -> 3.3V)
- MG21 A02011 B01VV1 2127
—> Zigbee module (on SM-011 V1.0 PCB)

Is it probably the serial flash memory because it supports SPI?

And do you maybe know where I can find the I2C pins on the PCB? I couldn’t find an easy position to connect additional I2C devices.

syssi, 2025-07-06:
Which hardware version do you own?

syssi, 2025-07-06:
There a two test pads called `IO33` and `IO32` on my hardware version:

[image]

```
i2c:
  - id: i2c_bus
    sda: GPIO33
    scl: GPIO32
    scan: true
```